adamb314/ServoProject #4: Brushed servos

Telepresent, 2022-04-12:
For building the robot you mention to use 3x coreless modified servos + 3x micro modified servos. Brushed servos are not required? Finding ATSAMD21G18A-A at a reasonable price for a custom PCB is very difficult this days. So, can we build the robot just with off the shelf boards?

adamb314, 2022-04-13:
That is correct, the current robot version does not use the modded DS3225. However, the MG959 can be replaced with DS3225 if someone wants to.

The reason behind the custom PCB is purely for convenience, so you can use Adafruit ItsyBitsy M0 Express boards instead if you want.

closed 2022-04-18 by adamb314

vkumarsinghnoida, 2022-04-29:
> That is correct, the current robot version does not use the modded DS3225. However, the MG959 can be replaced with DS3225 if someone wants to.
> 
> The reason behind the custom PCB is purely for convenience, so you can use Adafruit ItsyBitsy M0 Express boards instead if you want.

But will the program be the same and can we use any other board ?