adamb314/ServoProject #5: No current sensor？

pzh11001, 2022-04-24:
What a great project！
I noticed that previous versions had current sensors and corresponding control codes。
However, in the latest version, there is no current sensor on the circuit diagram. However, there is still a current sampling part in the code. Is that right?

adamb314, 2022-04-24:
Thank you! 

Yes that is correct. In the latest version I replaced the current controller with a sensorless model based approach, the CurrentControlModel class. Mainly to simplify the design but also because of the high noise levels of the current sensor. However, I left the old CurrentControlLoop class implementation if someone wants to change back to it for their design.

pzh11001, 2022-04-25:
> 
I'm sorry I didn't read it carefully enough
Oh, yes. I noticed that. Although there is code for  CurrentControlLoop, the program logic is not actually called. This program is called CurrentControlModel。 
According to my analysis of the code principle, the whole code actually has only three threads running independently: a main thread, a Controlloop thread of DCservo class, and a Statuslighthandler thread. The main servo control logic is executed in the Controlloop thread. However, since this is an ISR calling method based on timer interrupt, won't the execution of so much code in the interrupt function cause the CPU to get stuck？ because as far as I know, there should be very simple code in the ISR calling method.

adamb314, 2022-04-25:
No worries.

If it was a normal interrupt handler on the SAMD21 CPU you would be correct, but everything is taken care of by the ThreadHandler. If you want more information about it you could check out this: https://books.google.se/books?id=l9RtBgAAQBAJ&pg=PA673. Page 673 show the stack frame hacking I am using in platformSpecificClasses.cpp.

pzh11001, 2022-04-25:
> No worries.
>
> If it was a normal interrupt handler on the SAMD21 CPU you would be correct, but everything is taken care of by the ThreadHandler. If you want more information about it you could check [...]

Thank you. It's exciting. 
Now I want to use your project to make a small steering gear, and then make a small but high-precision 6axis robot arm.

pzh11001, 2022-04-25:
I consider that if a slight torsion spring is added to the output shaft, the gear backlash will always be eliminated in most cases. Although the backlash compensation algorithm is also needed in some cases, it can not happen too often. Because the compensation gap takes time, too free gap is easy to be frequently disturbed by environmental forces.
This should greatly improve the smoothness of the steering gear movement。
I'm going to test this device after reproducing your project

adamb314, 2022-04-25:
Sounds like a good plan. Good luck!

closed 2022-04-25 by adamb314