adamb314/ServoProject #6: Accuracy achievable

vkumarsinghnoida, 2022-04-25:
Hi Adam,
An awesome project. I was looking greater accuracy in servos for my automatic turret. Can you please tell me how accurate are the modified servos in degrees ?

adamb314, 2022-04-26:
Thank you.
With the latest version (using the the original potentiometer) I have been able to achieve 0.04 degree accuracy.

vkumarsinghnoida, 2022-04-26:
That's great sir. And can I modify my mg995 servo motor amd achieve the
same accuracy.

adamb314, 2022-04-26:
There are of course no guarantees... but it looks similar to the DS3225 I used so it should be possible. Good luck!

vkumarsinghnoida, 2022-04-26:
So what are the main things that a servo should have to follow your
tutorial. Like does the internal motor has to be the same or the size of
the servo should be the same ?

vkumarsinghnoida, 2022-04-29:
> Thank you. With the latest version (using the the original potentiometer) I have been able to achieve 0.04 degree accuracy.

So if I set the angle value to 90.04° will it be accurate or the value would be 90.04 (+/-) 0.04 ?

adamb314, 2022-04-29:
> I had problems with my servo motor so I opened it to solve the problem but I cannot understand where should I position the servo shaft and the pot shaft cause they both need to fit but I want the servos full range of positions. As you have opened so many servos how can I solve the problem ?
> 

I am sorry, but I don't really understand what you mean... Do you mean you want 360° of rotation?

> So if I set the angle value to 90.04° will it be accurate or the value would be 90.04 (+/-) 0.04 ?
>

it would be 90.04 (+/-) 0.04

vkumarsinghnoida, 2022-04-29:
I mean to say after I open the servo how do I position my potentiometer and servo shaft like how much should I turn or just fit them together ?

adamb314, 2022-05-01:
I just rotate the shaft until the keyway lignes up with the potentiometers shaft and then push them together.

vkumarsinghnoida, 2022-05-01:
Ok thank you very much.
Can you address my other question regarding the number of boards and the
code compatibility I have asked in another issue ?

closed 2022-05-01 by adamb314