adamb314/ServoProject #7: Alternative boards and code

vkumarsinghnoida, 2022-04-29:
Hi 
My questions are regarding how to make the servo without custom pcb as given in coreless servo build.
How many boards are required to control 2 servos?
How exactly do we connect the board and use which software ?
And can we use the original potentiometer instead of the magnetic encoder ?

adamb314, 2022-05-01:
1) One "Adafruit ItsyBitsy M0 Express" board per servo
2) a) If you don't want custom PCBs and use a potentiometer you should follow the Schematics in ServoProject/PcbDesignes/MicroServo/MainPcb/MainPcb.sch 
b) The same software
3) Using a potentiometer is the default when creating a new configuration

vkumarsinghnoida, 2022-05-02:
1)How can I find the gear ratio of my servo motor as it is not given in the
datasheet ? For the configuration program.

2)And which software do we need to boot the custom pcb and then do we just
run the python code or we need Arduino ide too ?

3)And how can I control it after I have finished all the steps without your
given gui ? Do I have to connect a usb cable or it needs pwm signal ?

adamb314, 2022-05-02:
1) best is to count the gear teeth to calculate it.
2) you program the board with the Arduino sketch.
3) please read "Communication bus setup for demo projects"

vkumarsinghnoida, 2022-05-03:
Thanks
Unfortunately I couldn't find communication bus setup code. Can you tell me
it's exact location (how to give the servo and angle)? And I have to
install sublime text to run the bootloader and other code ? Like what are
the steps from scratch.

adamb314, 2022-05-03:
You do not need to burn the bootloader if you use an off the shelf board, only when you have the custom PCB. I only use sublime text in the video to show text files and terminal output of the GUI.

Open one of the demos (there are Python, C++ and Arduino versions) to see how you control the servos.