adamb314/ServoProject #9: Compatibility with DS3218 servos

PJfixer, 2022-07-25:
Hi adam, 

First congratulations for your work, your results with theses cheap servos are really impressive

i'm verry interested in building five of them but before checkout on JLCPB, i would like to have your opinion about the PCB compatibility with DS3218 servos.

i think its ok because the only difference with 3225 servos seems to be height , and obviously the final torque.

what do you think ?

Regards,

adamb314, 2022-07-25:
Hi PJfixer, and thank you.
I agree with you, a DS3218 should work too. You might have to modify the height of the back cover though. Good luck with your servo modifications!

PJfixer, 2022-07-26:
great ! thank you ! 👍

closed 2022-07-26 by PJfixer