adamb314/ServoProject #14: Issue with running configurationWizard script on Windows

PointyyP, 2023-10-22:
I've installed MSYS2 and the PyGObject package, however an error "no such file or directory" occurs when trying to run the script.

PointyyP, 2023-10-22:
Thanks for the fast reply! Looks like fixing the directory helped. However, I have two new errors in the screenshot. 
[image]

smith901, 2023-10-22:
I got a notification for this issue and don't know why.   

After reading & looking at the error, seems like you need to install GTK3 on your machine: `Python/ServoProjectModules/GuiHelper.py:6` gi.require_version("Gtk", "3.0")

GTK does have MYSYS2 install instruction. I only use my Windows machine for offline business stuff. Hope this helps.

PointyyP, 2023-10-22:
It worked! Thank you so much!
[image]

closed 2023-10-22 by PointyyP