adamb314/ServoProject #15: Issue with bootloader

PointyyP, 2023-10-23:
After performing:

cd filename
sudo openocd

// I get "command not found"

[image]

smith901, 2023-10-24:
Looks like `openocd` isn't installed. Follow instructions [from this file in the project](https://github.com/adamb314/ServoProject/blob/main/PcbDesignes/ServoBrushed/burnBootloaderWithOpenOCD.txt).

PointyyP, 2023-10-24:
I was following the instructions and thought I installed openocd, but I'll give it a try again. Thanks!

PointyyP, 2023-10-24:
After installing openocd and soldering the three pins (D,C,Reset) I have a new error message after executing sudo openocd

[image]