adamb314/ServoProject #16: Maximum voltage of the brushed servo pcb

ShittyMetronome, 2023-11-03:
Hi !

Thank you for your incredible work, especially on making this project easy to make for everyone.

I have two questions regarding your custom pcb for brushed servos :
What is the maximum voltage allowed ? I would like to modify blue servos (ds3235 and ds3225) and power them with 2s batteries (8.4v max).
Does the modification change the max speed of the servo motor ? For example, I'd like to use blue ds3225 because of their speed (0.05s/60° @8.4v in theory).

Thank you !

Cydget, 2023-11-14:
Another option would be to change out the linear regulator with one that is rated for a higher voltage. You would need to make sure that the LDO/regulator does not exceed max power dissipation. Pd=(inputvoltage-3.3)*I_LDO. A good estimate for total LDO current is likely 100-150mA

ShittyMetronome, 2023-11-23:
Thank you both for your anwers !

closed 2023-11-23 by ShittyMetronome